rosmo-robot/Rosmo_ESC #3: Check clearance between screw terminals & M5 core

samuk, 2021-12-19:
We should check that the screw terminals have enough clearance when the M5 header is in place, stacked below an M5core or [Open Core ESP32](https://oshwlab.com/Rosmo/m5-proto)

runger1101001, 2021-12-19:
My plan is to use "pluggable terminal blocks" like the ODrive M5 module. I like that idea.

Normal terminal blocks won't work because the screws would not be accessible. And other connectors are either expensive, too bespoke, require crimping, etc, etc... the pluggable terminal blocks combine the ease of use and ubiquity of screw-connections with the "front access" only type of profile we need.

The pic here illustrates it nicely:
http://free-electronic.com/product/pluggable-terminal-block-2-pin/

So I have to find the ones we'll use (Amphenol I think, they're cheap) and I can send you the dimensions.

samuk, 2021-12-28:
Resolved

closed 2021-12-28 by samuk