rosmo-robot/Rosmo_ESC #4: Did you route the current outputs to mcu opamp capable pins?

samuk, 2021-12-19:
I noted you implemented the current sensors on each axis.

Did you route the current outputs to mcu opamp capable pins?

From Twitter: https://twitter.com/uuLipe/status/1472671409084243969

runger1101001, 2021-12-19:
Hmmm.. no I didn't pay attention to it. The driver has current sense amps built-in, so the signal is already amplified and in the right range for the ADC via the external resistors (if my calculations are correct!). What would the op-amps be used for?

uLipe, 2021-12-19:
Thanks for highlight my comment on twitter.

Sounds fair enough to me, I didn't noted the built-in amplification block on the driver.

samuk, 2021-12-19:
No worries, thanks for your interest. Would you be interested in testing the boards if we get a batch made?

closed 2021-12-19 by samuk

uLipe, 2021-12-20:
@samuk , for sure, I also have m5 stack esp32 modules here and it would be great to write an initial microROS network using these modules being an initial skeleton of an mobile robot.