rosmo-robot/Rosmo_ESC #7: Rosmo power supply concept?

runger1101001, 2021-12-27:
The power supply concept for Rosmo needs more attention.

The motors chosen are for 12V. They will run with less, but since they have quite high resistance the current, and hence the torque will be low.
m5stack power concept is very unclear to me. I can't find good documentation. [The schematic I can find ](https://github.com/m5stack/M5-Schematic/blob/master/Core/Basic/M5-Core-Schematic(20171206).pdf)has part of the power circuitry labeled "Speaker" :-( .

The general idea with m5stack seems to be that the core provides power to the rest, it has a 5V buck/boost converter with battery charging and 3.3V rails. It has a HPWR rail on the M5 header but it doesn't seem to connect to the core. The buck/boost converter is powered by the battery and the 5V_USB rail, and I assume will manage switching between these.

Currently the Rosmo ESCs 5V_USB rail connects into the 5V rail of the M5 header via a protective diode, so powering the core *should* would providing the Buck/Boost module doesn't mind this configuration (having its output back-powered).

An alternative could be to connect the 5V_USB to the VBAT on the m5 header, i.e. pretending to be a battery. Not sure if the buck/boost module will like that one either.

In any case the input to the ESC board should be at least 6V. The current design is for 4x1.5V battery, so this is just enough, but only when they're more or less charged up. Once they become empty the voltage would drop below 6V which is not ideal.

Note that the m5 core's battery charging is for 4.2V lipos or Lions, not for NiMH batteries in 4S. So I currently don't see how the m5stack itself is powered. The little battery in the base of the core module? – but the 3D design doesn't account for that...

Some ideas:

- get rid of the NiMH batteries in favour of a beefy 1S battery [like this](https://www.spystore.at/3-7v-2000mah-wiederaufladbarer-lithium-akku-e126849.htm) for example. This would get charged by the m5 core when plugged into USB, and could power everything for a little while. You'd need a boost converter board to boost it up to 12V for the ESC board input.
- use a 3S battery, something [like this](https://batteryzone.de/products/11-1v-2500mah-3s-hochleistungs-lithium-polymer-akku-fur-ferngesteuertes-auto-flugzeug-hubschrauber?variant=39892971192528&currency=EUR). There are other shapes and sizes. The 11.1V will be fine for the ESC and motor. Use a buck converter module to produce the 5V to power the m5 stack.
- use the NiMH battery pack, together with a boost module to get 12V and a buck/boost module to get stable 5V.

runger1101001, 2021-12-27:
Hmmm... this schematic for Core2 is a bit better: https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/schematic/Core/CORE2_V1.0_SCH.pdf

It shows clearly that HPWR isn't connected to the core. The other issues around VBAT 5V_BUS are the same.

samuk, 2021-12-27:
Just to clarify my current thinking is to use 3x 3.2v lifepo4 cells: https://m.aliexpress.com/item/1005002959027031.html

They share the same physical dimensions as Aa batteries.

runger1101001, 2021-12-28:
Ok, I will try to find space for it! I'm more optimistic about that than the CAN bus TBH. I can't really fit another port around the outside.

samuk, 2021-12-28:
On board canbus would be nice, but would also add cost. Don't we get similar functionality from plugging in a M5comm module for those use cases that need it?

runger1101001, 2022-01-07:
Ok, I've spent about 10h on the buck converter and I'm changing my mind on this:

- due to chip shortage none of the designs I've made before can be produced at the moment
- designing a new one based on available chips is a bit of work, and carries some risk - risk that it won't be right and messes up the rest of the board. Then we'd have to manually desolder stuff to be able to use the prototypes, I'd prefer to avoid that, really
- having it on a board with MCU and all the other stuff is also an expensive way to prototype the buck converter, which only needs a few components itself

So the new plan to is to release the board as it is, adding only a 3.3V regulator so it is self-sufficient when plugged into USB.
For the prototyping, we can use one of the many cheap and tiny buck regulator modules ("BECs") that are available on AliExpress or eBay. There will be space for it in the Rosmo body.
Then I can design a new buck converter for the Rosmo board v2, prototyping it separately, and then integrating it into the v2 design. At that point we'll also have a clearer idea what we want - maybe with battery charging? Or maybe more basic, just for the ESC board and seperating it from the whole complexity of the m5 power?

samuk, 2022-01-07:
Great, thanks for investigating! This sounds like a sensible way forward. I'm happy to spend a bit of time looking at existing open-source buck designs to see what we could make in the current chip situation.

Closing this, for now, we can create a new ticket when we want to look at this again.

closed 2022-01-07 by samuk

runger1101001, 2022-01-07:
No, the regulator is still missing and the m5 header... I'll export it again today if I can, but certainly before I order it.