rosmo-robot/Rosmo_ESC #10: Specify dimensions of encoder board

samuk, 2021-12-31:
As discussed it makes sense to develop a small encoder board so that users can plug and play the encoders. It would be good if this encoder could be used in a variety of motors.

[image]

I suggest we stick broadly to the above dimensions, whilst noting that the [BGM220](http://www.dys.hk/product/BGM2208-70.html) has 16mm spacing for the screw holes, so two 'notches' in the left and right of the above dimensions would give compatibility?

There would be a 7mm wide, side entry [JST SHR-06-VS ](https://www.jst-mfg.com/product/detail_e.php?series=231) without flange specified for soldering to the SPI side. This would be positioned 'in reverse' with the cabling going over the encoder chip so that it could be used in a [2205](https://ae01.alicdn.com/kf/H5db205d3acb24197b1f93c58d5f370b9e.jpg)

runger1101001, 2022-01-03:
Here a picture and 3D render:

[image]

[image]

This is the pinout:

[image]

These are the dimensions:

[image]

runger1101001, 2022-01-03:
If we can work with these for the prototype, then I have bunch here. I can provide AS5048As or TLE5012Bs in this shape.

If we want a different shape, we'd have to have some made :-)

samuk, 2022-01-03:
Great! nice one..

closed 2022-01-03 by samuk

samuk, 2022-05-07:
These are now available here: https://www.tindie.com/products/smallrobots/as5048a-encoder-board-for-robots-motor-control/