rosmo-robot/Rosmo_ESC #9: Export BOM

samuk, 2021-12-31:
It would be good to get a BOM, even if provisional and subject to change. It will enable us to get some ballpark production quotes.

runger1101001, 2022-01-03:
[Rosmo_ESC.csv](https://github.com/rosmo-robot/Rosmo_ESC/files/7804442/Rosmo_ESC.csv)

This is the current BOM.
This is the more or less finished board, but without the buck converter... I'm still playing with the design for that.

runger1101001, 2022-01-03:
I still need the part numbers for the M5 bus connector, top and bottom...

samuk, 2022-01-03:
Does this help? https://shop.m5stack.com/products/2x15-pin-headers-socket I can ask them on Twitter..

samuk, 2022-01-03:
Looks like you could specify the 3mm Samtec one? https://community.m5stack.com/topic/1980/pin-header-and-sockets-part-numbers

samuk, 2022-01-07:
In light of https://github.com/rosmo-robot/Rosmo_ESC/issues/7#issuecomment-1007344152 I think the BOM is final-v1 at this point?

runger1101001, 2022-01-09:
Hmmm... the Samtec one is 1.27mm pitch but the M5 bus is 2.54mm pitch. So that's not the right one. I'm beginning to to like this m5 stack so much, using custom connectors is just a pain.
If the grove port turns out to be a problem to procure I will change it to be a Stemma/QT I2C port instead.

I'm just going to assume standard 2.54 pitch SMD headers will work, the footprint looks like they will. We can ignore the alignment holes, I think, 30 solder joints should be strong enough for a few plug in/out cycles.

runger1101001, 2022-01-09:
Ok, this is more of an issue than I anticipated. Their custom connectors are also lower profile than standard ones. The lowest profile ones I've found so far, at mouser, have 3.05 or 3.6mm pins, but the standoffs are also taller than the m5 stack ones.
So you would end up with about 7.5-8mm distance between the board and the end of the pins.

But worse than that, these headers cost EUR5,50 each (!).

The smallest reasonably priced header will come to EUR0.70, but result in a board to pin distance of about 10mm.

I'll see if I can find anything else, at digikey for example... what a pain.

samuk, 2022-01-09:
If all else fails we could just buy the official one and have them shipped to whoever is doing the boards? https://shop.m5stack.com/products/2x15-pin-headers-socket

runger1101001, 2022-01-09:
I've included headers that should work - both sides for about $6, ugh. We can account for the difference in height in the ESCs shell, it will be a little taller module than necessary, but that could be a good thing anyway to get the heat out...

samuk, 2022-01-09:
Did you see these @75cents/ pc? https://shop.m5stack.com/products/2x15-pin-headers-socket

runger1101001, 2022-01-09:
I did look at that, but then I'd have to order them from m5stack, pay shipping+tax, send them back to china, pay shipping+tax, and by that time they wouldn't be any cheaper. And it would take an extra 1-2 weeks. Unfortunately, the forum already said PCBWay won't order from m5stack. And probably they wouldn't accept them for assembly anyway, since they're not on tape&reel.

Plus I don't really trust what I'd be getting. The picture says 10x set, the title says "4 pieces" and the description says "Includes 1xheader 1xsocket"...

I've put in the order for quotation... its a little more than I anticipated, they won't give the $30 assembly deal for some reason, prob. because it has 102 components or something. So for 10 boards + assembly it will come to $300 or so + the components, which still have to be quoted, but I'm guessing will come in at another $400 (counting the drivers and MCUs which I already have at around $20). 

So we're looking at somewhere around $70/board for a quantity of 10. Looking at the prices for making 50 pieces, it looks like the price would  be more like $10/board + components, so $30-$40/board.

runger1101001, 2022-01-09:
Yeah, its the component count that's driving up the price... but if I lower the number to 5 pieces to get the $30 deal, assembly is cheaper, but the PCBs cost is the same for 5 or 10 pieces... so the final cost is still around $30 per board. :-(
Still, we could save a bit by only producing 5, since we'll only need half the number of components.

samuk, 2022-01-09:
Shall we go for 5X? To start with? Split the cost between us and try and sell the 3x spare boards for $30 + P&P?

runger1101001, 2022-01-10:
Feedback from Fab so far: grove connector and m5 female header are unavailable...

I'll look for an alternative header, the female ones were not so hard to find.
I would suggest swapping out the grove connector for a JST 4-pin Stemma/QT style one? Or replacing it with just solder-points for a through-hole model?

runger1101001, 2022-01-10:
You mean leave the SMD footprint, and just leave it out? Sure, I could do that too. But there are also through-hole grove connectors, they'd be easier to solder.

samuk, 2022-01-10:
Yes, I understand your previous comment now, I'd assumed they were already through-hole. Let's do through-hole ones.

samuk, 2022-01-11:
https://store.open-electronics.org/2x15-Pin-Headers-Socket-2,54mm-Male-Connector-5330-M5CONN2X15M

runger1101001, 2022-01-11:
Those certainly look a lot cheaper!

But I have to stress that it isn't so easy as finding a web-link just anywhere. That doesn't look like a very well established store (don't know, just going by the website), and often fabs won't just order from anywhere - when they accept a order for assembly they take on responsibility for being able to source the parts incl. shipping, duties, etc... at the quoted price, have to meet the promised deadlines (in theory), and so won't take any risks... and that store is in Italy, so by the time those headers are at the fab in china you have add at least EUR30 in shipping, meaning for 10 pieces they're no cheaper than the ones we have.

And from the product page, it isn't clear it comes on a tape&reel with a pick&place compatible "pickup". The pick&place machines can't just grab a bare header, they need a kind of plastic cap on there.
If they don't have that the assembly has a manual placement step and things get more expensive that way.

Unfortunately you can't just send them a bag of components and say "assemble this". Its a bit more complicated.

runger1101001, 2022-01-11:
(I'm not trying to be negative, I'm just trying to explain why its not so easy. I strongly agree that a cheaper solution would be better, and that's something we should improve in V2).