rosmo-robot/Rosmo_ESC #11: Step/ Dir?

samuk, 2022-01-03:
I spotted that Odrive has Step/Dir https://docs.odriverobotics.com/step-direction

I know it's kind of crappy, but might be nice to support for 3Dprinters/CNC etc if not much work.

runger1101001, 2022-01-03:
We're running quite low on pins, but in particular board-space too... I'm just working on it, so I'll look into it.
Some options:
- we could support it using the m5header, but not give it its own port
- we could support it in software, i.e. via the SPI, I2C, UART or CAN inputs, and handle the step/dir impulses on the ESP32.
- we could drop support for something else, and add 2 x step-dir inputs connected to timer pins

My feeling is to leave this for the prototype version, and come back to it at a later time. Our principle goal is to create a solution for robots and not 3D printers. 3D printer users would prefer a different board, with support for 3 axes, inputs for end-stops, bed heating & fan outputs, etc...

samuk, 2022-01-03:
My feeling is to leave this for the prototype version, and come back to it at a later time. 

Great. Say no more!

closed 2022-01-03 by samuk