rosmo-robot/Rosmo_ESC #12: STM32G491MET6 availability

samuk, 2022-01-06:
So when @runger1101001 designed the board the parts were available. Unfortunately, that's no longer the case: https://octopart.com/stm32g491met6-stmicroelectronics-114199453#PriceAndStock

Farnell [claim some are coming in](https://uk.farnell.com/stmicroelectronics/stm32g491met6/mcu-32bit-170mhz-lqfp-80/dp/3648860) and hopefully they will, but other suppliers are showing a 6-12month lead time.

@uLipe floated the idea of an ESP32 version? I understand that you may not want to start from scratch & It may just be a case of waiting for parts to be available.

uLipe, 2022-01-07:
What would be the complexity to replace the stm32 to an ESP32 wrover -B module?

Please consider the ESP32 regardless the version has less GPIO than this STM32.

Otherwise it offers almost same peripherals or features.

runger1101001, 2022-01-07:
:-) the MCU shortage was anticipated. Indeed, the choice of MCU is not optimal, and is due to the shortage. I bought 20 pieces while they were one of the very few STM32s available at mouser, and the whole idea was to design the prototype for this MCU, because I have it on hand!
I can do kitted assembly at PCBWay to produce the prototype boards, and then there will be something to test, play with and make some initial videos of. :-)
Once this chip shortage situation improves, which looks to happen sometime during this year, we can produce more boards. And in the meantime we can design v2, based on the experience with the prototype, and targeting the right MCU that we want.
And the MCU we want is a cheaper, smaller STM32 in my opinion. The ESP32 is a great chip, but less well suited to motor control than the STM32s. It's ADCs are not as good, and the timers are less flexible. But the main problem is simply the number of pins. Even if you could use all the pins it would not be enough for 2 x motor + current sensing + 3 SPI buses + 2 x I2C, etc.... and the fact that you effectively can't use 6 of the pins because they're connected to the flash RAM means this chip is well suited to IoT use cases that use its built in WiFi, but not so much to this kind of task.

samuk, 2022-01-07:
Thanks, that clarifies the situation well I think. I have a few people who have expressed interest in the boards.

How would you plan to use the STM32 you have? 5pc run initially to test, then a v2. 

Or is it worth doing a 10pc run, keeping 10x STM32 in reserve?

runger1101001, 2022-01-07:
I would produce the max number possible under the inexpensive "prototyping deals" offered by the manufacturers. In my experience this is least expensive per board at low volume.
So at PCBWay, for example, this will be 10 boards, but the board has 100 components right now, and there can be other constraints like component limits which might mean they'll only make 5 or 6 boards for the low price.
I'll try to get a few quotes this WE, and order a first run.

samuk, 2022-01-07:
Great thanks, I think it would be good to get one into @uLipe's hands if possible.  I want one too Obvs :)

closed 2022-01-07 by samuk

uLipe, 2022-01-07:
Although ESP32 does not have the same good ADC than STM32 the chip would offer other features one of them is the fact of being a true dual-core chip, where is possible to left an entire core to control the motor tasks under FoC scheme ( and even in sensorless control), also there is an option of the MCPWM which is targeted to motion control able to generate by hardware the 6-pwm signals with configurable dead-time and hardware fault detection.

The problem I see here is the true miss of pins, which is being fixed on S3 model.

I'm just highlighting this because we're developing internally an motor control kit with a motion control component around s3.

Besides S3 or ESP32, what about using texas instruments C2000?

runger1101001, 2022-01-07:
Hey, the ESP32 is a great chip, and I have used it before in motor controller designs. I love the fact it has a IO MUX so much! It really is a great chip.
But please believe me that it can't handle 2 motors in 6-PWM + current sensing + SPI buses... 

SimpleFOC uses the MCPWM module to run the PWM on ESP32.

I look forward to trying out the S3, and maybe it is the right chip for the v2 design, lets see.

I have no experience of the C2000 architecture... is it well supported in Arduino? Unfortunately we don't currently support it in SimpleFOC, so one would have to write PWM and current sense drivers...