rosmo-robot/Rosmo_ESC #15: Stackable?

robottini, 2022-02-20:
Is the board stackable? Can I drive 4 motors?

runger1101001, 2022-02-20:
:-) Yes, it should be! I say _should_ because I'm waiting on the first test production, and have yet to try it out...

But the concept is to use I2C for comms, and there are 4 hardware-settable addresses, or you could set addresses in software. So the boards should stack.
In addition, they have a grove port (also I2C), so multiple stacks could be connected via cables, allowing a very large number of motors in theory.

Working against this concept is the limited space m5stack leaves for heat-sink or fan. So while I still have to try it out, I think much more than 2A will be unrealistic without a heat-sink of some kind, which will impose physical constraints on the stickability...

robottini, 2022-02-20:
thanks! I wait for a test!

closed 2022-02-20 by robottini